nakujaproject/N4-Recovery #64: Reparing the test stand

shohei, 2025-01-31:


PaulNKarimi, 2025-02-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

shohei, 2025-02-21:
Test stand fabrication is done. The ignition circuit is still being worked on.

shohei, 2025-02-21:
https://github.com/orgs/nakujaproject/projects/22/views/1?pane=issue&itemId=98883993&issue=nakujaproject%7CN-4_Solid_Propulsion%7C8

closed 2025-02-21 by shohei