nakujaproject/N4-Recovery #67: Consideration of safer test stand site

shohei, 2025-01-31:


NGGithua1, 2025-02-28:
We are in the process of designing a proposed test site, with ongoing consultations and design evaluations.

[Test site proj.pdf](https://github.com/user-attachments/files/19025735/Test.site.proj.pdf)
[Test Sitep.pdf](https://github.com/user-attachments/files/19025738/Test.Sitep.pdf)
[image]
[image]
[Test Site for Rockets.pdf](https://github.com/user-attachments/files/19025741/Test.Site.for.Rockets.pdf)
[image]
[TEST SITE 1.pdf](https://github.com/user-attachments/files/19025740/TEST.SITE.1.pdf)
[Test Site 2 BUDGET.docx](https://github.com/user-attachments/files/19025739/Test.Site.2.BUDGET.docx)
[Test Site 2 BUDGET.pdf](https://github.com/user-attachments/files/19025737/Test.Site.2.BUDGET.pdf)
[TEST SITE 2.pdf](https://github.com/user-attachments/files/19025736/TEST.SITE.2.pdf)

shohei, 2025-02-28:
# Suggestion 1
- Concrete blocks without foundation is better fro easy assembly/disassembly by students
## Tasks
- Check if the concrete blocks are sufficient to protect from explosion

# Suggestion 2
- Horizontal test stand can reduce the nessary height of the wall
## Task
- Check if we can change the direction of the test stand from vertical to horizontal. Fixing the test stand to the ground will be the major issue.

NGGithua1, 2025-03-07:
[image]

new design consideration.
The use of an excavator for digging is required for digging. 
Exact location for use needs to be identified.