nakujaproject/N4-Recovery #69: Modification of Launchpad

shohei, 2025-01-31:


jhnnyjava, 2025-03-14:
[image]

[image]

These are the new designs for the Launce pad base and legs, for the launch rail support, it will be 4.5 meters high so that the distance between it's tip and the launch rail tip is 0.5 meters. I'll be sharing the files is a few

NGGithua1, 2025-03-28:
Concept guiding the design

[image]
[image]

Ongoing launch pad design.

[image]
[image]
https://github.com/user-attachments/assets/d0bc761f-5170-4180-9cad-ac13e7ea0347

Incorporates a telescopic fitting of the rail support. 
Claculations and simulations pending.

NGGithua1, 2025-04-04:
[launchpad concept.zip](https://github.com/user-attachments/files/19602878/launchpad.concept.zip)
[N4 LAUNCHPAD DESIGN AND FABRICATION.pdf](https://github.com/user-attachments/files/19602879/N4.LAUNCHPAD.DESIGN.AND.FABRICATION.pdf)

shohei, 2025-04-04:
- Justification for load boundary condition. Why 523 N?
- Will required to have reinforcement structure. The result indicates the material strength limit is already passed.

Nellyndung-u, 2025-04-11:
1. Telescopic Design Simulation:

- Mass: 80.3 kg 
- Max Stress: 101.29 MPa (Less than 250MPa - yield strength of Mild Steel)
- Max Displacement: 3.48 mm 
- Safety Factor: 2.04

2. Single Bar Design:

- Mass: 86 kg 
- Max Stress: 14.64 MPa (very low compared to 250MPa - indicates overdesign)
- Max Displacement: 10.27 mm (Relatively high)
- Safety Factor: 14.14 (high - indicating overdesign)

Considering the design requirements to handle loading conditions with allowable stress of 250MPa and minimal displacement, the simulation results under:

- 4125 N load (applied on face to which the rocket rail will be attached) 
- A 5° tilt angle  
show that telescopic design is lighter (80.3 kg vs 86 kg) with lower displacement (3.48 mm vs 10.27 mm). However, it exhibits significantly higher stress (101.29 MPa vs 14.69 MPa) and a lower safety factor (2.04 vs 14.14) which is within the limit of 250 MPa. 
In contrast, single bar design is heavier (86 kg) with minimal stress (14.64 MPa), higher safety factor (14.14), though with higher displacement (10.27 mm). 

[CAD DRAWINGS.zip](https://github.com/user-attachments/files/19702435/CAD.DRAWINGS.zip)

[ANALYSIS REPORTS.zip](https://github.com/user-attachments/files/19702459/ANALYSIS.REPORTS.zip)

Nellyndung-u, 2025-04-12:
[LAUNCHPAD STEP FILES.zip](https://github.com/user-attachments/files/19720050/LAUNCHPAD.STEP.FILES.zip)

[DOWNWARD FORCE ANALYSIS.zip](https://github.com/user-attachments/files/19720051/DOWNWARD.FORCE.ANALYSIS.zip)

jhnnyjava, 2025-06-13:
[WEEK_2 Progress Report 2025 (1).pdf](https://github.com/user-attachments/files/20725841/WEEK_2.Progress.Report.2025.1.pdf)

Weeks two's progress