nakujaproject/N4-Recovery #72: Measure BMP sample rate and update frequency

bytecod3, 2025-02-05:


bytecod3, 2025-02-07:
Tested the BMP update frequency to check the speed of pressure and altitude update. Video link https://github.com/nakujaproject/N4-Recovery/tree/main/bmp-altitude-pressure-test

shohei, 2025-02-07:
@bytecod3 Describe your observation.

bytecod3, 2025-02-11:
When measuring the pressure and altitude, the BMP180 follows this sequence: Start temperature -> measure pressure -> Derive the altitude. See image below:

[image]

The BMP can be set to and advanced resolution mode that measures up to 128 samples per second. However, in the standard mode, the data rate is still high (takes around 7.5ms to convert pressure readings - see datasheet). This rate is enough for us. 

The effect this has on our altitude measurements is that they are fast enough to use on the rocket. I do not see any changes that need to be made. 

As a result I built a relative altimeter that measures relative altitude based on the starting point. And it works perfectly. The update frequency of the derived altitude is updated at 7.5ms

shohei, 2025-02-21:
The sampling frequency of BMP is 128Hz.

closed 2025-02-21 by shohei