nakujaproject/N4-Recovery #75: Dcoument data recovery procedures post-flight

bytecod3, 2025-02-07:


bytecod3, 2025-02-07:
Updates the data logging procedures
- Start logging after arming
- filename flight.txt
- docs updated here https://nakujaproject.com/N4-Flight-Software/index.html#step27
-

shohei, 2025-02-07:
@sirNindo Please review his code.

bytecod3, 2025-02-11:
@sirNindo I am waiting for your review .

sirNindo, 2025-02-21:
### ✅ What Works Well
Both SPIFFS(internal) & SerialFlash(external) Flash memories
**SPIFFS**

1.  SPIFFS Initialization (InitSPIFFS())

- [ ] ✔️ Correctly mounts SPIFFS. - in main.cpp
- [ ] ✔️ Uses FORMAT_SPIFFS_IF_FAILED = 1 to handle formatting failures.
- [ ] ✔️ Debug messages confirm success or failure.

2.  Logging & Reading Logs

✔️ logToFile() writes logs correctly based on the provided mode.
✔️ readLogFile() properly reads logs from SPIFFS.
✔️ getLogLevelString() maps log levels to human-readable strings.

### **❌** Issues & Required Fixes
InitSPIFFS()
To do 
 flag to show successful initialization use/implemetation -it is not fully utilized 

- [ ] 🔴 1️⃣ logToFile() – Fix File Clearing in Mode 0

Issue:
FILE_WRITE does not automatically clear the file before writing in mode 0. as intended

- There is no file to clear since it was not explicitly created ,,
- but if the computer had a test run , it would then exist and need formatting 

Fix:
Explicitly delete the file before writing:
In the logger.cpp
```sh
if (mode == 0) {
    if (fs.exists(file)) {
        fs.remove(file);  // Delete old log file before writing a new one(**_is missing_**)
    }
    File f = fs.open(file, FILE_WRITE);
}
```

- [ ] 🟠 2️⃣ logToConsole() – Missing Implementation

Was it intended to log the message to see the initialization statuses on a dashboard or just for debugging ?
Issue:
Declared in system_logger.h but not implemented in system_logger.cpp.
Fix:
Implement it in system_logger.cpp:
In the SystemLogger.cpp
The readLogFile() can be used in its placeand add functionality to log the serial prints over mqtt before launch
- as it may be used to ascertain initialization statuses 
or 
- maybe use boolean variables to show the status 

- [ ] 🟡 3️⃣ Improve Error Handling 

Issue:
Functions only print errors, but they don’t return success/failure to user.
Fix:

### ✅ Final Fixes Checklist
✔️ Fix File Clearing (delete file before writing in mode 0).
✔️ Implement logToConsole() (missing function in system_logger.cpp).
✔️ Improve Error Handling (return bool for logToFile()).
✔️ Handle "Missing File" in Append Mode (switch to write mode if file doesn’t exist).

### ** Proposed Steps**
🔹 Apply these fixes to ensure full functionality.
🔹 Test logging behavior in both write and append modes.
🔹 Confirm file clearing before flight logs start.