nakujaproject/N4-Recovery #77: Test Gps Signal range

sirNindo, 2025-02-07:


bytecod3, 2025-02-08:
Test the time to first-fix
Deliverables:
- Determine how long it takes to get the first fix at a given cloud coverage 
- Determine how fast the GPS updates itself on movement (e.g 100m) distance. Can it detect that?

sirNindo, 2025-02-16:
### Results 
**Time to first fix**

Time to first fix varies with cloud cover  Ranging from Half a minute to 5seconds -this is variable as when tested inside a building it takes longer and GPS satellites are specific.
The GPS( Neo 6MV2) would attempt connection to a satellite provided power is connected to it even without Serial.Read from it.

**How fast does the GPS update on movement?**
The update rate was 1 Hz after fixing to a satellite and updated after movement to about 2.5- 3.2  meters. 
The GPS would sometimes update location without movement which may be caused by noise due to obstructions and possibly the number of satellites locked onto. 
While the Neo 6MV2 might not be the most accurate GPS module, it can still provide decent precision. Even if they're a bit off from the true location, getting consistent readings can still be used as the data for our application where relative positioning is important.

### Recommendations 

- A delay in launch after powering the GPS would therefore give it time to fix to a Satellite 
- Good antenna: A high-quality antenna can help improve signal reception and thus, precision.-the existing antennae used is feasible 
- A clear view of the sky: Minimizing obstructions will lead to more consistent measurements.