nakujaproject/N4-Recovery #79: Discuss about the antenna (airframe team and recovery team)

shohei, 2025-02-07:


shohei, 2025-02-21:
@JeffreyGG001 Update the discussion.

JeffreyGG001, 2025-02-22:
[Altimeter Venting.pdf](https://github.com/user-attachments/files/18921549/Altimeter.Venting.pdf)

JeffreyGG001, 2025-02-22:
From our discussions, the following slots will be needed on the flight computer body tube section:
- 1 slot for the patch antenna wire - It may be easily taped to the body tube
- 1 slot for the remove before flight pin.
- 1 slot for the GPS antenna wire. 
- Of these slots, one may be made wider to allow for passage  of a USB debugging cable.
- Vent holes - A document discussing the sizing and placement of vent holes is attached below.

[Altimeter Venting.pdf](https://github.com/user-attachments/files/18921871/Altimeter.Venting.pdf)