nakujaproject/N4-Recovery #83: fix NaN on roll gyroscope roll calculation

bytecod3, 2025-02-09:


bytecod3, 2025-02-09:
There was a NaN result when calculating the pitch angle of the rocket. Pitch was calculated using arcsine formula which is known to produce NaN values if the operand is outside [-1, +1] range.

To fix this I clipped the operand to [-1, +1] range to make sure the operand of arcsine is always within the range

```c

double u = this->acc_x_ms / ONE_G;

    // clip to [-1, +1] bound before passing to arcsine
    if( ! ( (u > 1) || (u < -1) )) {
        this->pitch_angle = asin(this->acc_x_ms/ONE_G);
    }

```

bytecod3, 2025-02-11:
The NAN can be seen in the following packet. 

[image]

shohei, 2025-02-21:
Add else statement to add value in `this->pitch_angle`