nakujaproject/N4-Recovery #86: Implement remote arming of FC from ground station

bytecod3, 2025-02-11:


brian-io, 2025-02-11:
Hey, the arming is already implemented in the basestation here, file App.jsx line 48 onwards 
```js
 // Toggle arming state
    const newArmedState = !telemetry.operationMode;
    const message = new MQTT.Message(
      JSON.stringify({
        command: newArmedState ? "ARM" : "DISARM",
      })
    );
    message.destinationName = "n4/commands";
```
What's required is to prepare the topic "n4/commands"  in the FC to receive the "ARM" or "DISARM" command.
I've prepared a documentation of the various topics and what data is sent and received

bytecod3, 2025-02-12:
Okay. Well noted. Also @brian-io could you kindly make yourself available at IPIC ASAP for physical integration and testing of base station?

closed 2025-08-17 by khanji254