nakujaproject/N4-Recovery #89: Parachute testing with vehicle

shohei, 2025-02-12:
Martin and Nelly are working on it

Simulation:
https://github.com/shohei/n4-recovery-simulation

Matizkim, 2025-02-21:
TESTING JIG
Made of Mild steel, 
1. Angle line selected, sufficient to handle .
2. The tube: specific stress at the point of attachment to the loadcell.
3. The rod: fabrication requires much work and it will be overall heavier
DESIGN 
[image]
[image]
[image]
[image]
[image]

FABRICATION 
[image]
[image]

CIRCUIT 
The circuit used to be used is for the solids team

CODE
Initiator code that sends data form outside the vehicle
[Initiator.txt](https://github.com/user-attachments/files/18905788/Initiator.txt)

RESPONDER code that receives the data 
[RESPONDENT.txt](https://github.com/user-attachments/files/18905812/RESPONDENT.txt)

A python code that stores the data as CSV for future referencing and records real time graphical analysis from the corm port via the serial communication.
[DATA COLLECTION.txt](https://github.com/user-attachments/files/18905690/DATA.COLLECTION.txt)

SHOCK CORD
The shock cord 10m, is preferrable. A range of 8m - 12m has been used before. 10M cord provides a 6.54G, using it with extra knobs minimizes the G but not quantifiable yet.
[Parachute_Test_calculation.pdf](https://github.com/user-attachments/files/18905905/Parachute_Test_calculation.pdf)

TESTS
1. Test using the shock chord attached
2. Test without the shock chord

shohei, 2025-02-21:
@Matizkim Kindly relook at the formula of the deceleration.

[image]

NGGithua1, 2025-03-28:
Parachute test revised calculation

[Parachute.pdf](https://github.com/user-attachments/files/19504095/Parachute.pdf)

A common recommendation is that the shock cord should be 3 times as long as the overall rocket length.
Compared to an 8m shock cord, a 10m cord provides a lower deceleration rate of 10.86 G. 

chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://www.apogeerockets.com/education/downloads/Newsletter348.pdf

NGGithua1, 2025-03-28:
Next task is fitting of structure to the car, to be done from week starting 31st March 2025