nakujaproject/N4-Recovery #94: Documentation of Cooking Procedure

Ogweno-lab, 2025-02-13:


azid0azide, 2025-02-17:
## So...
We wrote down a procedure for cooking the Grain, but we ran into a small problem when the method wasn't followed to a tee.

We found that after cooking, removal of the core rods would be about 2-4 hours, but for the cooking on 17th Feb, there was a noticable oozing of not properly mixed and caramelized Sorbitol and 'waxiness' of the slurry after the wait time that was determined from previous cooking.

This was due to the modification of wait times and temperatures outlined in the procedure during the Cook of 17th.

azid0azide, 2025-02-17:
You can find the original write up [here](https://github.com/nakujaproject/N-4_Solid_Propulsion/blob/ikiara/Grains/Preparation/CookingKNSBTheRightWay.pdf).

shohei, 2025-02-21:
@azid0azide Upload the updated document.

azid0azide, 2025-04-19:
Here's the updated [document](https://github.com/nakujaproject/N-4_Solid_Propulsion/blob/main/Grains/Preparation/CookingKNSBTheRightWay.pdf)